morrownr/88x2bu #80: MTU cannot be set to more than 1500

backahasten, 2021-08-17:
MTU cannot be set to more than 1500
When configuring the ad-hoc based mesh network of Batman adv, you need to use the command:
**sudo ifconfig wlan0 mtu 1532**
However, I found that when the MTU is set to be greater than 1532, an error will be reported:
**SIOCSIFMTU: Invalid argument**
Is this a chip problem?

morrownr, 2021-08-20:
I did a little snooping around in the code to see what I could see. It appears 1500 is a hard coded limit in the driver. It isn't obvious why. I suspect making a change would result in breakage. Let me do a little reading about BATMAN so as to make a recommendation.

Can you give me an idea of your setup so I know what you are trying to accomplish?

backahasten, 2021-08-20:
I'm just trying to use 88x2 network card to build Batman adv network. You can https://www.open-mesh.org/projects/batman-adv/wiki Find more information.

backahasten, 2021-08-20:
It is an OSI layer 2 protocol, so MTU needs to be modified

morrownr, 2021-08-20:
After a little reading and testing...

BATMAN seems interesting. I may have to look harder at this.

I have experience with various USB WiFi adapters and drivers. My main box almost always has at least 2 usb wifi adapters connected. I decided to test the adapters I currently have attached:

$ sudo ifconfig wlan0 mtu 1532

$ sudo ifconfig wlan1 mtu 1532

$ ifconfig| grep -i MTU
enp2s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1532
wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1532

I did not see an error and the system is reporting the MTU of both adapters has increased to 1532. And everything is working well.

Now, what is the difference between what I tested and your adapter? Both of the adapters I tested are using the in-kernel Mediatek drivers (MT76). One adapter uses a mt7612u chipset (AC1200) and the other uses a mt7610u chipset (AC600).

You can find more information at the following site:

https://github.com/morrownr/USB-WiFi

Overall thoughts: My testing has shown that the 88x2 driver is not bad for managed mode. It has gaps in capability such as no WPA3 support but overall not bad. On the other hand. monitor mode and master mode support is not good at all. You would probably be well served by finding another job for your 88x2 adapter while getting an a adapter based on a mt7612u chipset for this BATMAN project. I can make a specific recommendation if you explain your use case.

Let me know how it goes.

backahasten, 2021-09-02:
Finally, I used Qualcomm's chip, and they seemed to do well. Many functions of WiFi chips are not commonly used, and Qualcomm has done a good job in these uncommon functions.

closed 2021-09-02 by backahasten

morrownr, 2021-09-02:
I agree. Qualcomm Atheros generally does a pretty job making sure less commonly used functions work well. The only problem is that we need them to released chipsets for the usb adapter market. If a project can use a card instead of an adapter, that opens up more options as wireless chipsets from Atheros, Mediatek and Intel are well supported in-kernel. Mediatek does a good job with less common features as well. I monitor the Linux Wireless mailing list.

I would avoid Realtek products. Period.