morrownr/88x2bu #81: ubuntu 20.04 kernel 5.11.0-27-generic

flare04, 2021-08-18:
The driver builds and installs successfully

output from iw dev
```
phy#8
	Interface wlxe84e068ad0a7
		ifindex 11
		wdev 0x800000001
		addr e8:4e:06:8a:d0:a7
		type managed
		txpower 42949572.96 dBm
```

Removed adapter
removed driver modprobe -r 88x2bu
```
Aug 18 10:17:07 jakku gnome-shell[2547]: Removing a network device that was not added
Aug 18 10:17:07 jakku wpa_supplicant[1561]: nl80211: deinit ifname=wlxe84e068ad0a7 disabled_11b_rates=0
Aug 18 10:17:07 jakku NetworkManager[1526]: <info>  [1629245827.3663] radio killswitch /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.2/ieee80211/phy8/rfkill13 disappeared
Aug 18 10:17:12 jakku systemd[1]: systemd-rfkill.service: Succeeded.
Aug 18 10:17:16 jakku kernel: usbcore: deregistering interface driver rtl88x2bu
Aug 18 10:17:27 jakku kernel: usb 1-2.1: new high-speed USB device number 20 using xhci_hcd
Aug 18 10:17:27 jakku kernel: usb 1-2.1: New USB device found, idVendor=0bda, idProduct=b82c, bcdDevice= 2.10
Aug 18 10:17:27 jakku kernel: usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 18 10:17:27 jakku kernel: usb 1-2.1: Product: 802.11ac NIC
Aug 18 10:17:27 jakku kernel: usb 1-2.1: Manufacturer: Realtek
Aug 18 10:17:27 jakku kernel: usb 1-2.1: SerialNumber: 123456
Aug 18 10:17:27 jakku systemd[1]: Starting Load/Save RF Kill Switch Status...
Aug 18 10:17:27 jakku mtp-probe[36214]: checking bus 1, device 20: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1"
Aug 18 10:17:27 jakku mtp-probe[36214]: bus: 1, device: 20 was not an MTP device
Aug 18 10:17:27 jakku systemd[2289]: Reached target Bluetooth.
Aug 18 10:17:27 jakku systemd[1]: Reached target Bluetooth.
Aug 18 10:17:27 jakku systemd[1]: Started Load/Save RF Kill Switch Status.
Aug 18 10:17:27 jakku kernel: Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8822
Aug 18 10:17:27 jakku kernel: Bluetooth: hci0: RTL: rom_version status=0 version=2
Aug 18 10:17:27 jakku kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_fw.bin
Aug 18 10:17:27 jakku kernel: Bluetooth: hci0: RTL: loading rtl_bt/rtl8822b_config.bin
Aug 18 10:17:27 jakku kernel: Bluetooth: hci0: RTL: cfg_sz 14, total sz 20270
Aug 18 10:17:28 jakku kernel: Bluetooth: hci0: RTL: fw version 0xab6b705c
Aug 18 10:17:28 jakku bluetoothd[1518]: Endpoint registered: sender=:1.88 path=/MediaEndpoint/A2DPSink/sbc
Aug 18 10:17:28 jakku bluetoothd[1518]: Endpoint registered: sender=:1.88 path=/MediaEndpoint/A2DPSource/sbc
Aug 18 10:17:32 jakku systemd[1]: systemd-rfkill.service: Succeeded.
Aug 18 10:17:38 jakku NetworkManager[1526]: <info>  [1629245858.0788] device (wlan0): driver supports Access Point (AP) mode
Aug 18 10:17:38 jakku NetworkManager[1526]: <info>  [1629245858.0793] manager: (wlan0): new 802.11 Wi-Fi device (/org/freedesktop/NetworkManager/Devices/20)
Aug 18 10:17:38 jakku systemd[1]: Starting Load/Save RF Kill Switch Status...
Aug 18 10:17:38 jakku kernel: usbcore: registered new interface driver rtl88x2bu
Aug 18 10:17:38 jakku mtp-probe[36231]: checking bus 1, device 20: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1"
Aug 18 10:17:38 jakku mtp-probe[36231]: bus: 1, device: 20 was not an MTP device
Aug 18 10:17:38 jakku systemd-udevd[36217]: Using default interface naming scheme 'v245'.
Aug 18 10:17:38 jakku systemd-udevd[36217]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Aug 18 10:17:38 jakku kernel: rtl88x2bu 1-2.1:1.2 wlxe84e068ad0a7: renamed from wlan0
Aug 18 10:17:38 jakku systemd[1]: Started Load/Save RF Kill Switch Status.
Aug 18 10:17:38 jakku NetworkManager[1526]: <info>  [1629245858.0995] rfkill15: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1:1.2/ieee80211/phy9/rfkill15) (driver rtl88x2bu)
Aug 18 10:17:38 jakku NetworkManager[1526]: <info>  [1629245858.1216] device (wlan0): interface index 12 renamed iface from 'wlan0' to 'wlxe84e068ad0a7'
Aug 18 10:17:38 jakku NetworkManager[1526]: <info>  [1629245858.1284] device (wlxe84e068ad0a7): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Aug 18 10:17:38 jakku systemd-udevd[36240]: Using default interface naming scheme 'v245'.
Aug 18 10:17:40 jakku ModemManager[1611]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1': not supported by any plugin
Aug 18 10:17:43 jakku systemd[1]: systemd-rfkill.service: Succeeded.
Aug 18 10:17:48 jakku systemd-udevd[36240]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Aug 18 10:17:48 jakku kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlxe84e068ad0a7: link becomes ready
Aug 18 10:17:48 jakku wpa_supplicant[1561]: nl80211: kernel reports: Authentication algorithm number required
Aug 18 10:17:48 jakku wpa_supplicant[1561]: P2P: No social channels supported by the driver - do not enable P2P
Aug 18 10:17:48 jakku wpa_supplicant[1561]: dbus: fill_dict_with_properties dbus_interface=fi.w1.wpa_supplicant1.Interface.P2PDevice dbus_property=P2PDeviceConfig getter failed
Aug 18 10:17:48 jakku NetworkManager[1526]: <info>  [1629245868.3571] sup-iface[0x5555b8ae3120,wlxe84e068ad0a7]: supports 5 scan SSIDs
Aug 18 10:17:48 jakku NetworkManager[1526]: <info>  [1629245868.3587] device (wlxe84e068ad0a7): supplicant interface state: starting -> ready
Aug 18 10:17:48 jakku NetworkManager[1526]: <info>  [1629245868.3588] Wi-Fi P2P device controlled by interface wlxe84e068ad0a7 created
Aug 18 10:17:48 jakku wpa_supplicant[1561]: wlxe84e068ad0a7: CTRL-EVENT-SCAN-FAILED ret=-1
Aug 18 10:17:48 jakku NetworkManager[1526]: <info>  [1629245868.3594] manager: (p2p-dev-wlxe84e068ad0a7): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/21)
Aug 18 10:17:48 jakku geoclue[2048]: WiFi scan failed
Aug 18 10:17:48 jakku NetworkManager[1526]: <info>  [1629245868.3602] device (p2p-dev-wlxe84e068ad0a7): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
[... 23 lines cut ...]
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     4B413DD09022C5FACE7D9C8
alias:          usb:v13B1p0045d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v20F4p808Ad*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p0115d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p0138d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2357p012Dd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0846p9055d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2001p331Cd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v2001p331Ed*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v7392pC822d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v7392pB822d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0B05p184Cd*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0B05p1841d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v13B1p0043d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApB812d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApB82Cd*dc*dsc*dp*icFFiscFFipFFin*
depends:        cfg80211
retpoline:      Y
name:           88x2bu
vermagic:       5.11.0-27-generic SMP mod_unload modversions 
```
Any ideas would be appreciated

morrownr, 2021-08-18:
What usb wifi adapter are you using?

What laptop are you using?

What exact model of Google WiFi are you using?

What module parameters did you set?

flare04, 2021-08-18:
Hi 

Laptop is a metabox Flo, which is based on Insyde corp L140CU
Adapter is a simplecom NW632
   |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 0bda:5411 Realtek Semiconductor Corp. 
        /sys/bus/usb/devices/1-2  /dev/bus/usb/001/003
        |__ Port 1: Dev 20, If 1, Class=Wireless, Driver=btusb, 480M
            ID 0bda:b82c Realtek Semiconductor Corp. 
            /sys/bus/usb/devices/1-2.1  /dev/bus/usb/001/020
        |__ Port 1: Dev 20, If 2, Class=Vendor Specific Class, Driver=rtl88x2bu, 480M
            ID 0bda:b82c Realtek Semiconductor Corp. 
            /sys/bus/usb/devices/1-2.1  /dev/bus/usb/001/020
        |__ Port 1: Dev 20, If 0, Class=Wireless, Driver=btusb, 480M
            ID 0bda:b82c Realtek Semiconductor Corp. 

I'll hunt out the exact model of Google Wifi, but it was the original version
I didn't change any from the install script 
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=2 rtw_beamform_cap=0

morrownr, 2021-08-18:
Try: rtw_switch_usb_mode=1

flare04, 2021-08-18:
OK changed as requested

Aug 18 14:52:04 jakku kernel: rtl88x2bu 1-3:1.2 wlxe84e068ad0a7: renamed from wlan0
Aug 18 14:52:04 jakku systemd[1]: Started Load/Save RF Kill Switch Status.
Aug 18 14:52:04 jakku NetworkManager[1526]: <info>  [1629262324.5345] rfkill17: found Wi-Fi radio killswitch (at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/ieee80211/phy10/rfkill17) (driver rtl88x2bu)
Aug 18 14:52:04 jakku NetworkManager[1526]: <info>  [1629262324.5538] device (wlan0): interface index 14 renamed iface from 'wlan0' to 'wlxe84e068ad0a7'
Aug 18 14:52:04 jakku NetworkManager[1526]: <info>  [1629262324.5596] device (wlxe84e068ad0a7): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Aug 18 14:52:04 jakku mtp-probe[52625]: checking bus 1, device 24: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Aug 18 14:52:04 jakku mtp-probe[52625]: bus: 1, device: 24 was not an MTP device
Aug 18 14:52:04 jakku systemd-udevd[52534]: Using default interface naming scheme 'v245'.
Aug 18 14:52:07 jakku ModemManager[1611]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3': not supported by any plugin
Aug 18 14:52:09 jakku systemd[1]: systemd-rfkill.service: Succeeded.
Aug 18 14:52:14 jakku systemd-udevd[52534]: ethtool: autonegotiation is unset or enabled, the speed and duplex are not writable.
Aug 18 14:52:14 jakku kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlxe84e068ad0a7: link becomes ready
Aug 18 14:52:14 jakku wpa_supplicant[1561]: nl80211: kernel reports: Authentication algorithm number required
Aug 18 14:52:14 jakku wpa_supplicant[1561]: P2P: No social channels supported by the driver - do not enable P2P
Aug 18 14:52:14 jakku wpa_supplicant[1561]: dbus: fill_dict_with_properties dbus_interface=fi.w1.wpa_supplicant1.Interface.P2PDevice dbus_property=P2PDeviceConfig getter failed
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7808] sup-iface[0x5555b8ae3120,wlxe84e068ad0a7]: supports 5 scan SSIDs
Aug 18 14:52:14 jakku wpa_supplicant[1561]: wlxe84e068ad0a7: CTRL-EVENT-SCAN-FAILED ret=-1
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7815] device (wlxe84e068ad0a7): supplicant interface state: starting -> ready
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7816] Wi-Fi P2P device controlled by interface wlxe84e068ad0a7 created
Aug 18 14:52:14 jakku geoclue[2048]: WiFi scan failed
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7818] manager: (p2p-dev-wlxe84e068ad0a7): new 802.11 Wi-Fi P2P device (/org/freedesktop/NetworkManager/Devices/24)
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7820] device (p2p-dev-wlxe84e068ad0a7): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7831] device (p2p-dev-wlxe84e068ad0a7): state change: unavailable -> disconnected (reason 'none', sys-iface-state: 'managed')
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7837] device (wlxe84e068ad0a7): state change: unavailable -> disconnected (reason 'supplicant-available', sys-iface-state: 'managed')
Aug 18 14:52:14 jakku NetworkManager[1526]: <warn>  [1629262334.7906] sup-iface: failed to cancel p2p connect: P2P cancel failed
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7907] device (wlxe84e068ad0a7): supplicant interface state: ready -> disconnected
Aug 18 14:52:14 jakku NetworkManager[1526]: <info>  [1629262334.7907] device (p2p-dev-wlxe84e068ad0a7): supplicant management interface state: ready -> disconnected
Aug 18 14:52:14 jakku wpa_supplicant[1561]: wlxe84e068ad0a7: CTRL-EVENT-SCAN-FAILED ret=-1
Aug 18 14:52:14 jakku geoclue[2048]: WiFi scan failed
Aug 18 14:52:15 jakku goa-daemon[2358]: secret_password_lookup_sync() returned NULL
Aug 18 14:52:16 jakku goa-daemon[2358]: secret_password_lookup_sync() returned NULL
Aug 18 14:52:17 jakku ModemManager[1611]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3': not supported by any plugin
Aug 18 14:52:17 jakku wpa_supplicant[1561]: wlxe84e068ad0a7: CTRL-EVENT-SCAN-FAILED ret=-1
Aug 18 14:52:17 jakku geoclue[2048]: WiFi scan failed

morrownr, 2021-08-18:
I don't have a clue at this point. What I do know is that these Realtek drivers do not support WPA3. In fact, if your wifi access point  has WPA3 or WPA3/WPA2 mixed selected, it won't work. Make sure 802.11w is not active in the access point either. Also, I'm pretty sure these Realtek drivers have never been tested on USB3 gen2 ports so that could be a factor.

There are reasons why I started the following repo:

https://github.com/morrownr/USB-WiFi

Realtek offers no public support. There is no known way to report bugs to Realtek. Realtek does not publicly release their drivers. It has been a long time since I have been able to find a new Realtek driver at a dealer web site. We may never see updated drivers from Realtek. 

My opinion is that Linux users should stay away from Realtek based USB WiFi adapters.

morrownr, 2021-08-19:
I know the above was a little harsh. Sorry about that. A little frustration showing. I am tired of Realtek not supporting their product. We can't fix everything out here in the community.

flare04, 2021-08-20:
No worries, thanks for taking the time to look at the issue.  I have raised it with the supplier, and pointed them to this ticket

closed 2021-12-05 by morrownr

terremoth, 2021-12-06:
@morrownr I did a lot of things trying to install the driver from this repo and from others, and after reboot "suddenly" worked, I don't know if it were from this repo or from another.
If this happens again, I will do it

middieNomad, 2022-01-08:
I stumbled across this github issue when trying to get my Simplecom NW632 usb adapter working on my arch installation. , Incidentally my usb adapter uses the same Realtek chipset mentioned  in this issue i.e. idVendor=0bda, idProduct=b82c, bcdDevice= 2.10. I managed to get the device working on my arch installation. Posting my steps here, perhaps this will help someone else.

```
sudo pacman -S linux-headers bc
git clone https://github.com/cilynx/rtl88x2bu
cd rtl88x2bu/
./deploy.sh
sudo modprobe 88x2bu
```

RRFreitas, 2023-07-11:
> I stumbled across this github issue when trying to get my Simplecom NW632 usb adapter working on my arch installation. , Incidentally my usb adapter uses the same Realtek chipset mentioned in this [...]

Thank you!!! This driver is the only one that worked here.