morrownr/88x2bu #82: Just wanted to thank you for making this...

blakeyjason, 2021-08-19:
I had been working on trying to debug a EAP-TLS issue between a Pi4 and OpenWRT for about a day and a half.  Then i had the bright idea to try an external dongle (which didn't have a driver available out of the box).   Used your scripts, build the driver, rebooted, and bang - EAP-TLS up and running.

Thanks very much!

morrownr, 2021-08-19:
I'm not a fan of Broadcom either. The good thing is the RasPi folks do give us the ability to deactivate it.

I like to tinker with RasPi's also. In case you are not aware of this repo:

https://github.com/morrownr/USB-WiFi

That repo has all kinds of information about USB WiFi on Linux/RasPi's and it has links to a lot of adapters that work with in-kernel drivers. The advantages of the in-kernel drivers are numerous... they are Linux Wireless standards compliant. WPA3 works... in fact, a lot of things work that won't work with these Realtek drivers.

Good luck with your project.

blakeyjason, 2021-08-20:
That page is super helpful!  Thanks for doing all that work.   Amazon order incoming...

Thanks again!

morrownr, 2021-08-20:
You are welcome. I like to see happy Linux users getting things done.

Now I'm curious what you ordered?

Tafhim, 2021-10-05:
I want to add to this instead of creating another. I had this Simplecom NW632 that was not working in neither Manjaro nor EndeavourOS (both are arch based). I don't know much other than `dmesg`. Used that info to google around and eventually found the repo. It just worked.

I have a few cases when it does have problems. Should I create a new issue to report those?

Also I am just curious how you debug issues like this? Is there a place where you start? I want to work on linux drivers myself so that I can contribute.

morrownr, 2021-10-06:
> I want to add to this instead of creating another. I had this Simplecom NW632 that was not working in neither Manjaro nor EndeavourOS (both are arch based). I don't know much other than `dmesg`. Used that info to google around and eventually found the repo. It just worked. I can't thank you enough.

You are welcome.

> I have a few cases when it does have problems. Should I create a new issue to report those?

Yes

> Also I am just curious how you debug issues like this? Is there a place where you start? I want to work on linux drivers myself so that I can contribute.

How about you start a new issue called "How do you debug issues?" and we'll see where it goes.

I'll be glad to help you learn to work on driver. Bring the topic up in the new issue.

Regards

closed 2021-12-05 by morrownr