morrownr/88x2bu #83: Your kernel headers for kernel 5.10.0-kali7-amd64 cannot be found

Nineplus99, 2021-08-19:
Dear Morrow,
i see your professional engineer, and when in install the drivers for the RTL8812BU, it come this error.

Could you help me to fix it?

Error! echo
Your kernel headers for kernel 5.10.0-kali7-amd64 cannot be found at
/lib/modules/5.10.0-kali7-amd64/build or /lib/modules/5.10.0-kali7-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
 
here is my step:
sudo su
lsusb
apt -y install git
apt -y install dkms
mkdir /apps
cd /apps/

--copy all the down lines to the command----  from the down, the error happed

git clone "https://github.com/RinCat/RTL88x2BU-Linux-Driver.git" /usr/src/rtl88x2bu-git
sed -i 's/PACKAGE_VERSION="@PKGVER@"/PACKAGE_VERSION="git"/g' /usr/src/rtl88x2bu-git/dkms.conf
dkms add -m rtl88x2bu -v git
dkms autoinstall

if you can help me, i am apprecite it.. i will give you a suprise.   my email:280200219@qq.com

morrownr, 2021-08-19:
Dear 9+99,

I am not a professional engineer but I'll try to help.

Step 1: Use the driver that is here. It has extensive installation documentation. You were using RinCat's driver. There is nothing wrong with RinCat's driver. He does a good job but if you need better documentation, use this driver.

https://github.com/morrownr/88x2bu

Step 2: There likely won't be a Step 2 if you do Step 1.

Nick

Nineplus99, 2021-08-19:
Dear Nick,
 Thanks so much. and i do by  your documentation,
-Installation Steps--
Step 3: Install the required packages (select the option for the OS you are using)
Option for Debian, Kali or Linux Mint Debian Edition (LMDE)

    $ sudo apt install -y linux-headers-$(uname -r) build-essential dkms git libelf-dev

Then it comes this ( E: Unable to locate package linux-headers-5.10.0-kali7-amd64)

udo apt install -y linux-headers-$(uname -r) build-essential dkms git libelf-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.10.0-kali7-amd64
E: Couldn't find any package by glob 'linux-headers-5.10.0-kali7-amd64'

So could you tell me know the reason of Unable to locate package linux-headers-5.10.0-kali7-amd64?  

Thanks you

John

morrownr, 2021-08-20:
John,

The problem is that the headers are not being installed. I will look at doing a test this afternoon but first:

Is this a recent installation of Kali?

Confirm that the computer is based on a amd64 processor?

morrownr, 2021-08-20:
My installation of Kali on amd64 shows the following for a kernel:

5.10.0-kali9-amd64

Try this:

$ sudo apt update

$ sudo apt full-upgrade

$ uname -r

Post the results of that last one.

morrownr, 2021-08-21:
John,

While I'm waiting on your reply to my post above, let me throw out some information.

Not many folks that run Kali use this driver. The reason is that Kali users are generally using apps for penetration testing and security analysis. These apps require solid, standards compliant monitor mode support. Monitor mode support is not that good with this driver. I and others have worked on it and it is better than what Realtek provided but it is not good.

The only modern Realtek driver that I consider reasonable to use with Kali is the driver for the rtl8812au chipset:

https://github.com/morrownr/8812au

With that said, my recommendation for Kali users is to use an adapter that uses in-kernel drivers. That will give you fully standards compliant support and the adapter will work according to provided instructions and you don't have to chase down drivers that won't compile (such as the case here.) A good list of recommended adapters and information can be found at the following site:

https://github.com/morrownr/USB-WiFi

Nineplus99, 2021-08-29:
Dear Nick  ,Thanks so much for your support. and  i upgrade the to 5.10.0-kali9-amd64. then it is ok now,   Thanks so much.
and  for this E: Unable to locate package linux-headers-5.10.0-kali7-amd64
E: Couldn't find any package by glob 'linux-headers-5.10.0-kali7-amd64'

i also find the headers and installed the headers  then it is also ok. thanks you so much. 

John

Nineplus99, 2021-08-29:
Dear Nick  
and my adater is Nineplus wifi adapter   and model No: wirelss wifi adapter ac1200. you can add it to the list.

Thanks so much

John

</body>
</html>

billserres, 2021-12-02:
> 

Where to find it this heater and in what steps? Please help me, I'm newbie here and I have similar ploblem.Thanks.

morrownr, 2021-12-02:
Hi @billserres 

This repo is depreciated. I need you to move to the new one:

https://github.com/morrownr/88x2bu-20210702

If you post this issue there, please include detailed information about what the problem is. I can't determine what the problem is from your post.

Regards

closed 2021-12-05 by morrownr

emreyaan, 2022-02-27:
> Hi @billserres
> 
> This repo is depreciated. I need you to move to the new one:
> 
> https://github.com/morrownr/88x2bu-20210702
> 
> If you post this issue there, please include detailed information about what the problem is. I can't determine what the problem is from your post.
> 
> Regards
I need help my kernel 5.10.0-kali9-amd64 but my kernel headers cannot be found I can't setup my new wifi adapter tp-link ac600 t2u plus but if I want to install linux-headers-5.10.0-amd64 : 
E: Unable to locate package linux-headers-5.10.0-kali9-amd64
E: Couldn't find any package 'linux-headers-5.10.0-kali9-amd64'
how can I fix this please help me

albatrosrkn, 2022-04-01:
> Author

 How do you upgrade the to 5.10.0-kali9-amd64? Please,Can you explain?

staticdev, 2022-04-02:
@albatrosrkn I honestly have no idea why you tagged me.

nos111, 2022-07-15:
I have found a solution and made a video explaining how to fix the missing header while installing this driver on kali
https://youtu.be/8A-scMfJDT8
Hope this helps

Saffa001, 2022-11-21:
Hello, Do you have any idea about my issue? 
Kali linux v 2022.4
6.0.0-kali3-amd64
I am trying to install the header and i am getting bellow error message.

apt install -y linux-headers-6.0.0-kali3-amd64 build-essential bc dkms git libeif-dev

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libeif-dev

I am really apreciate your advise.

Thank you.

Nineplus99, 2022-11-22:
Dear friend

i think you can update the code and then all will be ok

john from ninepluswifi.com

morrownr, 2022-11-22:
> Hello, Do you have any idea about my issue? Kali linux v 2022.4 6.0.0-kali3-amd64 I am trying to install the header and i am getting bellow error message.
>
> apt install -y [...]

Hi @Saffa001 

This repo is only a link to the currently active repo. Please go to:

https://github.com/morrownr/88x2bu-20210702

Since I don't know what repo this question is about. my recommendation is:

$ sudo ./remove-driver.sh
Delete your driver directory.
Start installation again using the repo above.

This issue will be deleted after this message is sent.

Nick