morrownr/88x2bu #86: Cannot compile on CentOS7 3.10.0-1160.42.2.el7.x86_64

wimein, 2021-09-19:
I can't get it complied on my CentOS 7 box:

[root@lukas 88x2bu]#  ./install-driver.sh 

Running install-driver.sh version 20210725
Starting installation.
Copying source files to: /usr/src/rtl88x2bu-5.8.7.4
Copying 88x2bu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl88x2bu/5.8.7.4/source -> /usr/src/rtl88x2bu-5.8.7.4

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=3.10.0-1160.42.2.el7.x86_64 KSRC=/lib/modules/3.10.0-1160.42.2.el7.x86_64/build......(bad exit status: 2)
Error! Bad return status for module build on kernel: 3.10.0-1160.42.2.el7.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.

[root@lukas dkms]# uname -a
Linux lukas.box 3.10.0-1160.42.2.el7.x86_64 #1 SMP Tue Sep 7 14:49:57 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
[root@lukas dkms]# lsusb
...
Bus 003 Device 002: ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU]
...
[root@lukas dkms]# 

###############################
/var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log

[make.log](https://github.com/morrownr/88x2bu/files/7191623/make.log)

##############################

Please help ...

morrownr, 2021-09-19:
Hello wimein,

Thanks for the report. I took a look at make.log and the other information you provided. My conclusion: I need to convince you to replace CentOS7 with a supported OS. Kernel 3.10 went End of Life in 2017. To work on this issue I would need to install CentOS7 and I simply do not have time to do that given everything else that is on my to-do list. As a general rule, I do not work on issues that come up with kernels that are End of Life.

If you would be so kind as to tell me what software you intend to use, what CPU, amount of RAM and amount of storage that your computer has, I can make a recommendation for a more modern Linux distro for you to install.

Regards,

Nick

wimein, 2021-09-19:
Hi Nick,

Thank you very much for your reply. The 3.10 kernel is indeed old, but CentOS 7 is still supported and will be until 2024. As we are quite happy with CentOS we consider moving to CentOS 8 Stream. But even the current CentOS 8 kernel version (4.18.0) has reached end of life, so it seems there is no chance to get your driver running on any existing recent CentOS distribution? It would be great if your driver could be compiled on any of the recent CentOS family members.
I use my machine for development purposes privately, but I manage a large number of CentOS boxes (> 100) at work. 

Regards, Winfried

morrownr, 2021-09-19:
Winfried,

I am pondering the situation. Let me throw out some thoughts to see what might be a good option in your eyes:

1) If CentOS 7 is the preferred OS, maybe we should look at the usb wifi adapters that are supported in that kernel. You have an AC1200 adapter but is that your real requirement? N150? N300? N600 (dual band)? AC600? AC1200? I haven't checked but I suspect that kernel has internal support for up to N600. Tell me what your wifi requirement is and we can explore the options. Yes, it is very possible to buy adapters that go back a few years and many are available that are supported by in-kernel drivers. In fact, I maintain a site all about adapters supported with in-kernel drivers:

https://github.com/morrownr/USB-WiFi

2) I am most familiar with the Debian world. I keep test partitions with Manjaro and Fedora so as to test but Debian is where my expertise is. Is there a version of CentOS that is based on kernel 4.4? The reason that I am asking is that 4.4 is a Super Long Term Support kernel (possibly up to 2036). The oldest kernel that I test with is 4.9 which is in the x86 Raspberry Pi Desktop OS. This driver works well with 4.9 so I suspect we would be getting close if 4.4 is an option for you.

Installing CentOS 7 and getting this driver going on it would take far more time than I have available but maybe I can help you find a solution.

dkmcode, 2021-09-20:
You could consider running a 5.4 or 5.14 kernel from the elrepo repository ( http://elrepo.org/tiki/HomePage ). The package names are kernel-lt and kernel-ml respectively. I've been using the elrepo kernels for years. It's hard to compile modules against the CentOS 7 kernels because of all the patches Redhat does to the kernel

morrownr, 2021-09-20:
>  It's hard to compile modules against the CentOS 7 kernels because of all the patches Redhat does to the kernel

That explains a lot. The make.log was a mess. 5.4 is a LTS kernel so I would recommend it for a test if you decide to try the above suggestion. If the 5.4 kernel in the elrepo repo is generic or close to it, you might very well see a clean compile.

dkmcode, 2021-09-20:
> If the 5.4 kernel in the elrepo repo is generic or close to it, you might very well see a clean compile.

Every time I have looked at the package source, they have been generic kernels and your code compiles cleanly against them. I have been using your code with the elrepo LTS kernel as the access point on my home router for a while now (since I got tired of trying to patch another Realtek adaptation to compile myself and found this one).  Thanks for all your work by the way.

morrownr, 2021-09-21:
@dkmcode 

Hey, I appreciate you stopping by to help the OP. Hopefully we can help him find a solution.

Glad that this driver is working well for you. It keeps me busy working on this driver and 4 more plus the USB-WiFi repo.

wimein, 2021-09-21:
Thank you both for help and support.
Installing kernel-lt from elrepo does indeed help. On CentOS 7 compiling the driver while running this kernel still results in errors. However, on a CentOS 8 Stream machine with kernel-lt installed and running, the driver can be compiled without warnings/errors and loaded into dkms.

WLAN on that machine does work. Authentication via MSCHAPv2 against some Linux RADIUS server succeeds (tunneled TLS/MSCHAPv2/no certificate), authentication against a Windows 2019 RADIUS server succeeds, as well (authentication via PEAP/MSCHAPv2/no certificate).

Thank you very much again.
best regards, Winfried

dkmcode, 2021-09-21:
I forgot one important step for Centos 7. You need to install gcc 9. If you search for and install the packages: devtoolset-9-gcc devtoolset-9-binutils devtoolset-9-runtime, and activate it before compile with
`. /opt/rh/devtoolset-9/enable`
It will compile. The Centos 7 gcc 4.8.5 is not new enough to compile a 5.4 kernel or any module you are trying to compile against it

dkmcode, 2021-09-21:
You can find the gcc 9 packages at https://centos.pkgs.org/7/centos-sclo-rh-x86_64/

dkmcode, 2021-09-21:
Or
`yum install centos-release-scl
then the packages you need should be available through yum.

wimein, 2021-09-21:
Compiling the driver on CentOS7 using gcc 9 sounds great, but unfortunately doesn't fully work. 

`[root@ms01 88x2bu]# yum install -y centos-release-scl
...
[root@ms01 88x2bu]# yum install -y devtoolset-9-gcc devtoolset-9-binutils devtoolset-9-runtime
...
[root@ms01 88x2bu]# scl enable devtoolset-9 bash
[root@ms01 88x2bu]# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/opt/rh/devtoolset-9/root/usr/libexec/gcc/x86_64-redhat-linux/9/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,lto --prefix=/opt/rh/devtoolset-9/root/usr --mandir=/opt/rh/devtoolset-9/root/usr/share/man --infodir=/opt/rh/devtoolset-9/root/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --with-default-libstdcxx-abi=gcc4-compatible --enable-plugin --enable-initfini-array --with-isl=/builddir/build/BUILD/gcc-9.3.1-20200408/obj-x86_64-redhat-linux/isl-install --disable-libmpx --enable-gnu-indirect-function --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
gcc version 9.3.1 20200408 (Red Hat 9.3.1-2) (GCC) 
[root@ms01 88x2bu]# uname -a
Linux ms01.bnmrz.org 5.4.147-1.el7.elrepo.x86_64 #1 SMP Thu Sep 16 08:39:39 EDT 2021 x86_64 x86_64 x86_64 GNU/Linux
[root@ms01 88x2bu]# 

[root@ms01 88x2bu]# ./install-driver.sh 

Running install-driver.sh version 20210725
Starting installation.
Copying source files to: /usr/src/rtl88x2bu-5.8.7.4
Copying 88x2bu.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl88x2bu/5.8.7.4/source -> /usr/src/rtl88x2bu-5.8.7.4

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.4.147-1.el7.elrepo.x86_64 KSRC=/lib/modules/5.4.147-1.el7.elrepo.x86_64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.147-1.el7.elrepo.x86_64 (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.8.7.4/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
[root@ms01 88x2bu]# 
`

[make.log](https://github.com/morrownr/88x2bu/files/7204186/make.log)

The machine is fully updated and is booted into kernel-mt kernel.

morrownr, 2021-09-21:
Whew. That is a really messy make.log file. The pain level is growing. Let me reiterate an earlier suggestion.

https://github.com/morrownr/USB-WiFi

That repo has information and links to around 70 usb wifi adapters that are supported by in-kernel drivers. ALL adapters at that site are supported from kernel 4.19 on. There are adapters at that repo that are supported with kernel 3.10.

Here is the deal. If Red Hat's patches had triggered compiler problems with the in-kernel drivers, those doing the patches would have known immediately and would have taken appropriate action.

On the other hand, those doing the patches would not know that they are blowing up these Realtek drivers because these Realtek drivers are out-of-kernel. Additionally, these Realtek drivers are not Linux Wireless standards compliant and that is a BIG DEAL! I put warnings here in these Realtek repos warning Linux users that these Realtek repos are here to help those that are new to Linux and want to keep existing Realtek adapters going and I STRONGLY recommend users to not buy Realtek adapters. The USB-WiFi repo has more info on this subject.

I'll turn rant mode off now.

dkmcode, 2021-09-21:
Those errors are consistent with compiling with gcc 4.8.5.  From the module source directory try
`. /opt/rh/devtoolset-9/enable
make -j4 KVER=5.4.147-1.el7.elrepo.x86_64
`
see if you get a clean compile. I have included the logs from a compile with gcc 4.8.5 and one that works with 9.3.1

[compile-gcc-4.8.5.txt](https://github.com/morrownr/88x2bu/files/7205288/compile-gcc-4.8.5.txt)
[compile-gcc-9.3.1.txt](https://github.com/morrownr/88x2bu/files/7205292/compile-gcc-9.3.1.txt)

wimein, 2021-09-22:
@dkmcode: 

Enabling gcc 9 with
`. /opt/rh/devtoolset-9/enable`
and compiling with
`make -j4 KVER=5.4.147-1.el7.elrepo.x86_64`
does work without errors.
It also works if I enable the newer gcc with
`scl enable devtoolset-9 bash`
I do however consistently get compilation errors when I use the install_driver.sh script.
It looks like `dkms build ...` cannot be convinced to use the newer gcc, at least I don't know how to enable gcc 9 in dkms. There are people trying to replace gcc by clang in dkms build on debian. I added "/usr/bin/scl enable devtoolset-9 bash" to /root/.bashrc, but the "dkms build .." still throws errors.

@morrownr : Kudos for a great summary of supported devices, I probably would have been better off to buy a different stick.

dkmcode, 2021-09-22:
I have never used the install_driver.sh script or the dkms system. If you don't want to install the module by hand on each kernel update, I am using the akmods system which automatically recompiles the module and creates and installes a new rpm package containing the module on each kernel update. If anyone is interested, I will clean up the SPEC file and post it here.

wimein, 2021-09-23:
This is what I've done to make this work:

yum install -y git centos-release-scl kernel-lt kernel-lt-devel --enablerepo=elrepo-kernel
yum install -y devtoolset-9-gcc devtoolset-9-binutils devtoolset-9-runtime

reboot into 5.4.148-1 kernel

mkdir ~/WLAN
cd ~/WLAN
git clone https://github.com/morrownr/88x2bu.git
cd 88x2bu
scl enable devtoolset-9 bash
make -j4 KVER=5.4.148-1.el7.elrepo.x86_64
make install
modprobe -r 88x2bu
modprobe 88x2bu

reboot into 5.4.148-1 kernel.

WLAN does work as expected.

So in summary, I could successfully compile the 88x2bu driver on CentOS 7 and also get it running with the kernel-lt kernel.

Thank you again for your support!

closed 2021-12-05 by morrownr