morrownr/88x2bu #88: How to cross-compile for OpenWrt?

Coniface, 2021-10-03:
I have been trying to cross-compile this driver for OpenWrt for days, also pardon me for my inexperience.

I downloaded OpenWrt SDK which includes the toolchain to cross-compile packages for it. If I'm correct, I'd simply have to run from your repository `make` with KSRC (the kernel sources directory) and CROSS_COMPILE (the cross-compilation tool from the toolchain) set in environment, right? I cannot seem to find the kernel as I always get an error saying there is no modules directory.

Should I rather be looking in OpenWrt's SDK to build the driver as a package for OpenWrt?

Also I read an issue of someone who built the driver for a Raspberry. Thing is I cannot find and install dkms on my router (NanoPi R4S) like on a Raspberry so I necessarily have to do it "the hard way" (cross-compiling).

morrownr, 2021-10-04:
> I have been trying to cross-compile this driver for OpenWrt for days, also pardon me for my inexperience.

Don't apologize for inexperience. You appear to be fixing the inexperience.
 
> Should I rather be looking in OpenWrt's SDK to build the driver as a package for OpenWrt?

I don't know. I use OpenWRT on a router here but I'm just a user, not a developer. My dev efforts go into the drivers and info here:

https://github.com/morrownr?tab=repositories

You may not have noticed that I have drivers up for all modern dual band Realtek chipsets and info for the Mediatek chipsets. That does give a somewhat broad perspective on the capabilities. With that said...

there may not be much info out there that applies to the 8812bu chipset. This chipset is pretty good for managed mode but would not be my first choice for AP mode. Until we recently stumbled onto a fix that really helped AP mode, the AP mode in this driver was not good at all. It is better now but I would lean toward adapters based on these chipsets:

mt7612u
mt7610u
rtl8812au
rtl8811au

The first two chipsets already have drivers as a OpenWRT package:

kmod-mt76x2u      MediaTek MT76x2U wireless driver (for the mt7612u)

You can find a lot of information about available adapters at this repo:

https://github.com/morrownr/USB-WiFi

Not the answers you were looking for but maybe it helps.

Coniface, 2021-10-04:
Yes I had already checked all your respositories and especially your long explanations about Atheros and Mediatek adapters being better than Realtek in terms of compatibility. But I had already bought the adapter some time ago, also I tried to give it a look but I could not find adapters with kernel-ready drivers that have dual-band 1200Mbps and AP mode. Also, I find difficult to find adapters with specific chipsets, seems that sellers are not always keen to specify the chipset.

Thanks anyway, maybe someone else will find that issue and know how to do it? In the meantime, I'll try giving it a shot and come back if I manage to solve it.

morrownr, 2021-10-04:
What you may want to do is post a msg over in the OpenWRT forums. That is where the OpenWRT experts hang out.

I understand that it can be hard to find adapters with specific chipsets but it can be done. I have a lot of experience doing such and I might be able to help in that regard if I know what city and country you live in.

I understand about already having a Realtek adapter. I have some also. I'm just trying to do the best I can to keep well maintained drivers for usb wifi available for Linux users. While I'm at it, I think it would be wrong to not point out how well the Mediatek based adapters compare and there is a big difference based on the difference in the drivers.

I had a few extra minutes today so I plugged on of my mt7612u based adapters into the wifi router that runs OpenWRT. I installed the driver package and it was up and running in no time. Now I have 2 5 GHz networks on the router, in addition to the 2.4 GHZ network. It was fun.

dmitttri, 2021-10-29:
Hi, 
despite the fact that AP mode with this driver is bad and unreliable, I can try to provide you help with cross compilation process. I am not really familiar with OpenWrt, but steps are pretty much standars, and SDK independent.

First make sure toolchain works properly on your system. You may need to add toolchain "bin" folder to yout PATH. Then try to compile a simple "hello world" C program, and check executable format of your compiled program, with "file" command.

Then make sure you have properly compiled your kernel with this toolchain. To compile kernel, you'll have to go to the SDK kernel source, and run something like "make ARCH=arm64 CROSS_COMPILE=aarch64-linux-". 

And then finally, you can compile your driver. This is example form my system:

make ARCH=arm64 CROSS_COMPILE=aarch64-linux- KVER=4.4.213-gfeb5427d-dirty KSRC=/home/fcast/work/kernel-rk3328/kernel-4.4.213-gfeb5427d-dirty

I hope this may be helpful for someone else stuck with cross copmilation,

It it may be of any help, with your cross compilation process, this is the line I am using to cross-compile this driver for arm64 architecture for my Rockchip boards:

make ARCH=arm64 CROSS_COMPILE=aarch64-linux- KVER=4.4.213-gfeb5427d-dirty KSRC=/home/fcast/work/kernel-rk3328/kernel-4.4.213-gfeb5427d-dirty

closed 2021-12-05 by morrownr