morrownr/88x2bu #89: no 5GHz anything

Musemagic, 2021-10-04:
Hi,  First, thank you for making this and keeping it updated.

It works perfectly on the 2.5GHz frequency and nothing on the 5GHz, no matter what I set it to or my router 5GHz band to, mode, channels, etc.

USB dongle works perfectly in Win10.  

Device is:  MiiLink Wireless USB WiFi Adapter for PC/Desktop,1200Mbps Dual Band 2.4GHz/5GHz High Gain 5dBi Antennas USB 3.0 Wireless Network Adapter 

chipset:  Realtek 8812BU

OSes:  Linux Mint 19.3, amd64
        Debian 11 (nonfree distro

tried on both USB 2.0 and USB 3.0.  

I tried rtw_vht_enable=2

I also tried various standards, 802.11ac, 802.11n/g mixed, overlapped channels, etc. just no recognition of a 5GHz band that I can see.

morrownr, 2021-10-04:
There is only one situation when you should use ```rtw_vht_enable=2``` and that is when you are using the adapter in 5 GHz master (AP) mode and I didn't see you saying anything about AP mode so please return that setting to the default but going into your driver directory and running: ```$ sudo ./edit-options.sh```

When you are editing that file, please post the entire options line for me to look at once you are finished editing and have saved the file.

Musemagic, 2021-10-05:
Sorry, I mean to imply I changed the options back as well.   Options are default (and left as default):
# Edit the following line to change options:
options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=2 rtw_beamform_cap=0

Also, kernel in Debian 11 is 5.10.08 so two different kernals, amd64.

morrownr, 2021-10-05:
There is a section in the README that goes over suggested wifi router setting. You might crosscheck that.

Have you previously installed another driver?

Ypu can change the following to 1 if you put the adapter in a usb3 port:

rtw_switch_usb_mode=2

Musemagic, 2021-10-05:
Yes, I changed my router channels for 5gHz  I use WPA2.  

More telling,  in Windows 10, everything worked perfectly, which implies it is not HW failure or my router, settings.  

I already changed the rtw_switch_us_mode=2 is for USB 3.0 and I did change it back when in a USB 2.0 hub.  

There is simply no 5gHz or 802.11ac.  

I have a hard time believing that my router settings need to be different for a Linux driver but I already tried many different settings and I use open source firmware, dd-wrt,, which enabled a lot of options not available in proprietary firmware.  

dmesg, iwconfig, nothing shows a 5gHz active channel.  It appears from the bitrate, in 2.4GHz, 144Mb/s, 802.11n is working.

morrownr, 2021-10-05:
> I already changed the rtw_switch_us_mode=2 is for USB 3.0 and I did change it back when in a USB 2.0 hub.

The file called 88x2bu.conf is where the documentation for the parameters is located.

rtw_switch_us_mode=2 sets USB2
rtw_switch_us_mode=1 sets USB3

In case this is not clear, To makes changes to the driver settings, you do not modify the 88x2bu.conf in the driver directory. The copy in /etc/modprobe.d is the one to change:

$ sudo nano /etc/modprobe.d/88x2bu.conf

> There is simply no 5gHz or 802.11ac.

Had you tried to install another driver prior to installing this one? One of the leading causes of problems is users that have tried to install another driver which did not work but messed their system up without them knowing it. I keep an information repo up. It provides a lot of information that is good for Linux users to know:

https://github.com/morrownr/USB-WiFi

Please run and post the results of the following?

$ sudo dkms status
 
> I have a hard time believing that my router settings need to be different for a Linux driver

That would make sense. But as one goes through life, one discovered more and more things that don't make sense.

I understand that you have a problem. Stay calm and work with me and we will likely find the problem. Something that may not be obvious to you is that this driver repo currently averages about 2,000 hits per week. To say that this is a popular site and the driver is heavily used is an understatement.  It is probable that if this was a problem with the driver, we would most likely see hundreds of reports in issues... but we don't so it is likely we will find something specific to your location that is the cause. 

Regards

Musemagic, 2021-10-06:
dkms status:
rtl88x2bu, 5.8.7.4, 5.4.0-87-generic, x86_64: installed

 I did not modify the 88x2bu.conf file, I simply copied the line for the purposes of this comment area.  I have already set the rtw_switch_us_mode, I also tested it in both Mint 19.3 and Debian 11 as previously stated. 

I did install other drivers but I removed them and they are gone..  Interesting is that cilynx also had the same result, no 5GHz, yet I know on the USB device, it works and exists in Windows 10 so had to be a Linux/driver issue.  Standard debug, the system is the only change, i.e. Linux/driver.     This device, it's a Chinese knockoff 
lsusb for device ID gives:  0dba:b812

morrownr, 2021-10-06:
> dkms status: rtl88x2bu, 5.8.7.4, 5.4.0-87-generic, x86_64: installed

> I did install other drivers but I removed them and they are gone.

dkms status confirms that you cleaned out the other drivers.

>. Interesting is that cilynx also had the same result, no 5GHz,

cilynx does a good job.

> ...it works and exists in Windows 10 so had to be a Linux/driver issue.

Or it could be a user issue.

> I think I've found out the issue, I have assumed this driver supports 802.11n, 802.11ac @ the 5GHz bandwidth.

That is a good assumption because this driver does indeed support 802.11ac.

> I finally set my router to issue mixed 802.11a/n @ 5GHz and viola, works with USB 2.0 port used speeds of n (144MB/s).

Is your router capable of 802.11ac?

> am I wrong to think this driver supports 802.,11ac dual band @ 5GHz?

You are correct to think this driver supports 802.11ac. In fact, this driver is VERY fast in managed mode (client) when an uncongested 80 MHz channel width 5 GHz channel is available. It can exceed 400 Mb/s speed.
 
> DOH! I was reading that 2.4GHz (g/n) and 5GHz (a) forever on the settings, gee, maybe it meant what it said! (ha) In my defense, i802.11a is so old, like 802.11b, I just didn't register that was the actual protocol!

My best guess is that the problem is with router settings or router capability. I use and am familiar with OpenWRT so I could be of help there but I have never used dd-wrt.