morrownr/88x2bu #96: May be worth calling out the importance of `rtw_switch_usb_mode` in the README

skylarmb, 2021-10-22:
After fiddling with this repo not working for me (I could see wifi networks but was in an infinite hanging/connection loop state), i finally randomly tried `rtw_switch_usb_mode=1` and instantly everything worked perfectly.

Considering the README recommends USB 3, should `rtw_switch_usb_mode=1` be the default, or at least get a mention in the USB section?

morrownr, 2021-10-23:
Ouch! You are right, this should probably get a write up in the USB recommendations section of the README.

The current default (2) is the result of problems with the other two options (0, 1). In fact, 2 has worked very well for a long time. It is interesting how this USB stuff is handled manually by Realtek because I also use adapters based on Mediatek chipsets. The Mediatek drivers, which are in-kernel, don't offer a module parameter. This is fully automated. I've never had a problem and have never seen anyone report a problem related to USB with the Mediatek drivers.

Anyway, it is what it is for now. Could I get you to write a paragraph for me to include? Your perspective would likely help you do a good job.

Regards

eddiegb, 2021-10-30:
@morrownr

Thanks for confirming it's a known issue. However, it is not your bad. If it's anyone's bad it's Realtek's, for not engaging with the their Linux users and leaving us completely on our own to figure these things out about their products, firmware & drivers. You, on the other hand, have gone out of your way to help us all. Whatever you might have omitted to do is irrelevant when you've freely given us so much already, despite being under no obligation.

closed 2021-12-05 by morrownr