morrownr/88x2bu #104: Error! Bad return status for module build on kernal: 5.11.0-43-generic (x86-64)

jthanp, 2021-12-16:
Spinning up a new Ubuntu 20.04 (focal) box, and pulled over a usb wifi adapter I've used on an 18.04, but getting an odd exception when trying to install:  

`Building module:
cleaning build area...(bad exit status: 2)
'make' -j8 KVER=5.11.0-43-generic KSRC=/lib/modules/5.11.0-43-generic/build...(bad exit status: 2)
ERROR (dkms apport): binary package for rtl88x2bu: 5.13.1 not found
Error! Bad return status for module build on kernel: 5.11.0-43-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
`
Any ideas on how to correct this issue?

morrownr, 2021-12-16:
Hi @jthanp 

This has come up before and I don't think the issue was solved. I just did some searching and I see a lot of links related to this. I read a few but have not come to a conclusion. Here is what I think I know:

- The issue seems to center around Ubuntu 20.04 and distros based on 20.04.
- The error is not in this driver.
- The issue affects other drivers such as video drivers that need to be compiled so it isn't just a wifi driver problem.
- My main dev box is Linux Mint which is based on Ubuntu 20.04 but I have never seen this problem.
- I am running kernel 5.13 based on the hwe kernel that can be installed with this package:

linux-generic-hwe-20.04-edge

- The real issue seemed to be related to `ERROR (dkms apport)` as I am seeing people use the manual installation with success:

$ sudo ./remove-driver.sh
$ make clean
$ make
$ make install
$ sudo reboot

The manual installation does not use dkms so that tends to indicate it is a problem with dkms. I have an installation of Ubuntu 21.10 that I use for testing and I have not seen this so there is that option as well.

Lastly, I really need you to move this issue over to ...

https://github.com/morrownr/88x2bu-20210702

...because this repo is just a placeholder these days. I just happened to see this issue but this repo is not monitored.

Nick

jthanp, 2021-12-21:
Nick, thanks for your reply.  I'm not sure how to move this question to the 20210702 repo, I tend to lurk on github, but participate only when my hair's on fire :).
I tried the manual build, but when I try to make or make clean, I get a message like *** No rule to make target 'clean'.  Stop.
Typically when I see this I look for a configure script, but I'm not seeing one of those either.