morrownr/88x2bu #109: Raspberry pi 4 - Archer T4U vs Archer T3U

jsalatiel, 2022-10-21:
Hi, I am trying to make this module work on my raspberry pi4 running libreelec.
I have two wifi dongles, **TP-Link Archer T3U** and **TP-Link Archer T4U ver.3**.
Technically both should work using this driver, but I just can not make the T4U work.

Here are the logs for both:

dmesg for   TP-Link Archer T3U:
```
[  561.892462] usb 1-1.1: new high-speed USB device number 8 using xhci_hcd
[  561.982994] usb 1-1.1: New USB device found, idVendor=2357, idProduct=012d, bcdDevice= 2.10
[  561.983004] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  561.983009] usb 1-1.1: Product: 802.11ac NIC
[  561.983014] usb 1-1.1: Manufacturer: Realtek
[  561.983019] usb 1-1.1: SerialNumber: 123456
[  563.144811] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```
wifi scanning working just fine:
```
#  iw dev wlan0 scan
...
```

Now for the T4U:

dmesg:
```[  644.635817] usb 1-1.1: new high-speed USB device number 9 using xhci_hcd
[  644.726432] usb 1-1.1: New USB device found, idVendor=2357, idProduct=0115, bcdDevice= 2.10
[  644.726451] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  644.726461] usb 1-1.1: Product: 802.11ac NIC
[  644.726471] usb 1-1.1: Manufacturer: Realtek
[  644.726480] usb 1-1.1: SerialNumber: 123456
[  644.864374] usb 1-1.1: 88XXau (null) hw_info[d7]
```
ifconfig still shows the wlan0 interface, but:
```
# iw dev wlan0 scan
command failed: Network is down (-100)
#  ifconfig wlan0 up
ifconfig: ioctl 0x8914 failed: Operation not permitted
``` 

modinfo output
```
#  modinfo  88x2bu | head -2
filename:       /lib/modules/5.10.95/RTL88x2BU/88x2bu.ko
version:        v5.13.1-20-gbd7c7eb9d.20210702_COEX20210316-18317b7b
```

lsusb output
```# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 9, If 0, Class=Vendor Specific Class, Driver=rtl88XXau, 480M
```

Any idea about what may be causing this?

morrownr, 2022-10-21:
Hi @jsalatiel 

I need to see the device ID of both adapters. Can you put them both in a port and post the results of:

$ lsusb

jsalatiel, 2022-10-22:
$ lsusb
Bus 001 Device 004: ID 2357:012d TP-Link Archer T3U [Realtek RTL8812BU]
Bus 001 Device 003: ID 2357:0115 TP-Link Archer T4U ver.3

morrownr, 2022-10-22:
Well, unfortunately you are right. Both adapters do use the 8812bu chipset. That makes the diagnosis easy. These Realtek driver don't support more than one adapter with the same chipset in the same system. I address this in the FAQ. Actually you should be posting this issue in the active repo:

 https://github.com/morrownr/88x2bu-20210702

If one of your AC1200 adapters used a rtl8812au chipset, you would be in business but I usually recommend users go to the Main Menu and select item2 which gives a long list of adapters that use in-kernel drivers (plug and play.) The driver is in the kernel. Here is the link to the Main Menu:

https://github.com/morrownr/USB-WiFi